PaulStoffregen/Encoder #3: Adding ESP8266 Compatibility

acropup, 2016-01-12:
Hi Paul,

First: You are awesome and I'm super impressed by all you've done for the community. I love the Teensy 3.1 and it is my go-to for prototyping. I'm excited to have something to contribute back to you. :stuck_out_tongue_closed_eyes:

I wanted to use a rotary encoder on an ESP8266 (I'm using a NodeMCU 0.9), and I have it working now with pin interrupts. I'd like your opinion on where some of the code changes should go. In [interrupt_pins.h](https://github.com/PaulStoffregen/Encoder/blob/master/utility/interrupt_pins.h), you write:
```
// Teensy (and maybe others) define these automatically
#if !defined(CORE_NUM_INTERRUPT)
```
I don't know how standard it is for a board to make these definitions, or for libraries to use them. Should I propose to the [ESP8266](https://github.com/esp8266/Arduino/) repo that we add those defines, or should we just add them to this library? They're all pretty straightforward, as far as I can tell. I've read that all GPIO except GPIO16 support pin interrupts, but I'll have to look more into that, as the datasheets I've found don't get into that sort of detail. It'll look something like this:
```
[...]
// ESP8266 (https://github.com/esp8266/Arduino/)
#elif defined(ESP8266)
  #define CORE_NUM_INTERRUPT EXTERNAL_NUM_INTERRUPTS
  #define CORE_INT0_PIN		0
  #define CORE_INT1_PIN		1
  #define CORE_INT2_PIN		2
[...]
```
In the process, I also found a small bug affecting the ESP8266 portInputRegister() definition, which I'll look at hopefully getting fixed before updating [direct_pin_read.h](https://github.com/PaulStoffregen/Encoder/blob/master/utility/direct_pin_read.h).

PaulStoffregen, 2016-01-12:
> Should I propose to the ESP8266 repo that we add those defines, or should we just add them to this library?

Either is fine

acropup, 2016-01-13:
Ok, I think I'll add them to this library. Regarding the ESP8266 Arduino bug, it has been [addressed](https://github.com/esp8266/Arduino/issues/1110) and is staged for the 2.1.0 release. I don't know how far away that will be. Would you rather I submit the code that will be compatible with 2.1.0, or should I submit the workaround that works with 2.0.0? It should continue to work into 2.1.0, but it bypasses some preprocessor macros, which is in theory a problem, but is fine with the current supported hardware.

acropup, 2016-01-16:
Feature added by pull request
https://github.com/PaulStoffregen/Encoder/pull/4

closed 2016-01-16 by acropup