PaulStoffregen/Encoder #7: Encoder::update is private with optimized interrupts

romainreignier, 2016-02-09:
Hello Paul,
First of all, thanks a lot for this great library.
I was using the library for some time on Arduino UNO and MEGA and it works well, but I have missed the **ENCODER_OPTIMIZE_INTERRUPTS** feature. I have then tried to enable it by adding the following define statement before including the library:
```C
#define ENCODER_OPTIMIZE_INTERRUPTS
```
But I get an error while building it:
```
Encoder::update(Encoder_internal_state_t*)' is private
```
In fact, the `update` method is called in the interrupt routines so it has to be available publicly. Replacing the `private` statement [l. 184](https://github.com/PaulStoffregen/Encoder/blob/master/Encoder.h#L184) by `public` was sufficient to be able to compile and use the sketch.

It seems that this bug is present in the whole Github history and I am a bit surprised that nobody has reported it yet. So if I missed something, forgive me and close the issue. Or maybe most people, like me, do not read all the doc and do not use this feature.

Tested with Arduino 1.6.5, 1.6.7, library versions 1.3 and 1.4.

Sincerely,

Romain

romainreignier, 2016-02-15:
Also referenced here : <https://forum.pjrc.com/threads/32986-Optimized-Interrupts-for-Encoder-library-not-working-after-update>

PaulStoffregen, 2016-02-27:
This one is easy to fix, but it looks like the newest versions of Arduino use a different linker command which might still cause problems.  Ugh...

closed 2016-02-27 by PaulStoffregen

romainreignier, 2016-02-29:
Thanks :+1:

commit e05a3243 (2016-02-27): Fix update for ENCODER_OPTIMIZE_INTERRUPTS
fix #7