PaulStoffregen/Encoder #8: attachInterrupt redefinition with optimized interrupts

romainreignier, 2016-02-10:
Hello,
When using the **ENCODER_OPTIMIZE_INTERRUPTS** statement, the file `utility/interrupt_config.h` is used.
In this file, the following macro [l.6](https://github.com/PaulStoffregen/Encoder/blob/master/utility/interrupt_config.h#L6) is used to redefine the `attachInterrupt` method:
```C
#define attachInterrupt(num, func, mode) enableInterrupt(num)
```
The issue is that it interferes with other uses of this method name like with the [TimerOne & TimerThree libraries](https://www.pjrc.com/teensy/td_libs_TimerOne.html). In these libraries, the `attachInterrupt` method only takes one argument but as the previous macro is not scoped, it is used as well in the Timer libraries. So the compiler throws that kind of errors:
```
Arduino/libraries/TimerThree/TimerThree.h:141:39: error: macro "attachInterrupt" requires 3 arguments, but only 1 given
     void attachInterrupt(void (*isr)()) __attribute__((always_inline)) {
                                       ^
```

A simple workaround would be to replace the `attachInterrupt` method in the Timer libraries by something like `attachInterruptHandler` or `attachInterruptFunction` but it will break the API and Arduino users are already used to that name.

But the issue is caused by the Encoder library so it would be better to fix  that in this library.

Sincerely,

Romain

romainreignier, 2016-02-14:
Oh, I am so sorry, I missed the warming on the library page.
I close the issue.

PaulStoffregen, 2016-02-27:
I'm looking at this now....

closed 2016-02-27 by PaulStoffregen

romainreignier, 2016-02-29:
Thanks :+1:

commit ed5ff8cf (2016-02-27): Don't interfere with attachInterrupt
fix #8