PaulStoffregen/Encoder #12: Encoder with esp-07: Fatal exception 0 (Illegal Instruction Cause)

brunnwart, 2016-04-24:
Dear Paul,

I use your routine and get random fatal exceptions. After spending unsuccessfull hours may I ask you for help?
I tried Arduino IDE 1.6.[7,8 and latest build] and all options in the menu for flash/cpu mode or frequency, different ESP-07, all with the same result: the function crashes randomly, although I use yield().
Any help would be greatly appreciated - I'm running out of ideas :-(

Thanks a lot and regards,
Ditmar

--
Error message (and my function used is cited below):

Fatal exception 0(IllegalInstructionCause):
epc1=0x402027c0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Exception (0):
epc1=0x402027c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0: ........

void vorzurueck(int swtch, int mtr12pwm, long enc_steps) { 
  boolean anhalt = mcp.digitalRead(swtch);
  if (!anhalt) {
    encmotor1.write(0);
    long tmp_enc = 0;
    mcp.digitalWrite(mtr12pwm, 1); 
    while (!anhalt && tmp_enc < enc_steps) { 
      anhalt = mcp.digitalRead(swtch);
      tmp_enc = encmotor1.read();  
      tmp_enc=abs(tmp_enc);
      yield();
      if (tmp_enc %1000==0) Serial.println(tmp_enc);
    }
    yield();
    mcp.digitalWrite(mtr12pwm, 0);
  }
}

PaulStoffregen, 2016-04-24:
I can't help with ESP

brunnwart, 2016-04-24:
Seems like it esp-07 does not crash / reboot when
#define ENCODER_DO_NOT_USE_INTERRUPTS
is used before
#include <Encoder.h>
I do not know yet if it is loosing counts with that option.

brunnwart, 2016-04-25:
Unfortunately #define ENCODER_DO_NOT_USE_INTERRUPTS is not an option as (the esp doesn't crash but) the encoder looses a lot of counts with it :-(

fredericplante, 2016-06-28:
You better off asking your question on the Arduino/Esp8266 branch brunnwart.

By the way, Paul Your SerialFlash library is pretty much 100% compatible with the Esp8266/arduino platform, now, by the way. It was only requiring 2 little changes, on in the SDtoflash sketch and one in the ESP8266 version of the SPI.h file, that does not really concern you any way..

Great work! We might take a look at the audio part of your work, that all is really great, thank you so much.

realholgi, 2017-02-02:
On ESP8266 interrupt routines you need to prefix with ICACHE_RAM_ATTR to avoid crashes.

See my pull request https://github.com/PaulStoffregen/Encoder/pull/15 or pull from my fork at https://github.com/realholgi/Encoder

vks007, 2017-09-09:
Hi @realholgi , I used your files but I am still getting exceptions in the code. It did not help.
I get exceptions even if I don't use interrupts. The illegal instruction exception line reported in the stack trace in Encoder.h is:
	static void isr12(void) { update(interruptArgs[12]); }
I am using ESP8266 ....

realholgi, 2017-09-09:
@vks007 obviously you defined ENCODER_USE_INTERRUPTS , which I have not used fixed. Try to undef it or also prefix the corresponding line 774 with ICACHE_RAM_ATTR...

vks007, 2017-09-09:
Hi @realholgi , sorry I did not understand your suggestion. Can you be a bit more specific please. 
fyi, I am using your libraries & not from Paul.