PaulStoffregen/Encoder #17: Add support for nRF52 boards

rtyley, 2017-04-16:
Can I offer 0.03 bitcoin (~$40) towards making a new release of your Encoder library with support for nRF52 boards?

Specifically, I'm trying to run an Arduino sketch on a [Adafruit Feather nRF52 Bluefruit](https://www.adafruit.com/product/3406) - there's no helper chip (eg `ATmega32u4` or `ATSAMD21`) with this product, instead code is run directly on the [`nRF52832`](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52832).

Having successfully used your Encoder library with other boards, I've had a couple of compilation errors compiling for the`nRF52`:

* libraries/Encoder/utility/interrupt_pins.h:202:2: `error: #error "Interrupts are unknown for this board, please add to this code"`
* libraries/Encoder/Encoder.h:61:11: `error: 'IO_REG_TYPE' does not name a type`

...new values need adding to [`interrupt_pins.h`](https://github.com/PaulStoffregen/Encoder/blob/master/utility/interrupt_pins.h) & [`direct_pin_read.h`](https://github.com/PaulStoffregen/Encoder/blob/master/utility/direct_pin_read.h), but I'm inexperienced in this area and even after a bit of digging around the board definition in [adafruit/Adafruit_nRF52_Arduino](https://github.com/adafruit/Adafruit_nRF52_Arduino) I still wasn't sure what to add :disappointed:

If you could make a new release of your library with this support I'd appreciate it - if necessary, you can put the Bitcoin towards purchase of a Feather nRF52 for testing!

PaulStoffregen, 2017-04-16:
Sorry, I'm not available for a new project, even on a paid consulting basis.  I also do not have experience with this particular hardware.

Perhaps you can find someone who knows this hardware and is willing to do the work and submit a pull request?  These register defs are fairly easily understood by anyone who's an expert with the particular hardware.

rtyley, 2017-04-17:
Fair enough! I've asked for help on the Adafruit forum- Adafruit [have replied](https://forums.adafruit.com/viewtopic.php?f=57&t=115655&sid=e3c4705d0cca70f7ed9266c8fe93f798#p577848) that they'll put it on the list of things to work on, but the nRF52 is a new board and it'll be a while before they get to it.