PaulStoffregen/Encoder #18: Can't use interrupt pins on Arduino/Genuino 101

DaisukeUra, 2017-06-15:
Hi there.

I use this library on my Arduino/Genuino 101.
When only use "change" supported interrupt pins, such as 7&8, Encoder instance looks like doesn't count pulses.
It is properly counting when I use "change" not supported pins.
How I use "change" supported interrupt pins on this library?

Best regards.

DaisukeUra, 2017-06-15:
I try this code, looks like properly counting.

```
#include <Encoder.h>

Encoder *myEnc;

void setup() {
  Serial.begin(9600);
  Serial.println("Basic Encoder Test:");
  myEnc = new Encoder(7, 8);
}

long oldPosition  = -999;

void loop() {
  long newPosition = myEnc->read();
  if (newPosition != oldPosition) {
    oldPosition = newPosition;
    Serial.println(newPosition);
  }
}
```

I don't know why it works, but my problem was maybe solved.
But I think it remains some problem in the library's code.

DaisukeUra, 2017-06-15:
I've try this code and it properly works.

```
#include <Encoder.h>

Encoder *myEnc;

void setup() {
  Serial.begin(9600);
  Serial.println("Basic Encoder Test:");
  myEnc = new Encoder(7, 8);
}

long oldPosition  = -999;

void loop() {
  long newPosition = myEnc->read();
  if (newPosition != oldPosition) {
    oldPosition = newPosition;
    Serial.println(newPosition);
  }
}
```

I don't know why it works well.
I think it remains some problem in library's code.

Thanks.

PaulStoffregen, 2017-06-15:
https://www.arduino.cc/en/Main/ArduinoBoard101

"change is only supported by pins 2, 5, 7, 8, 10, 11, 12, 13"

closed 2017-06-15 by PaulStoffregen

DaisukeUra, 2017-06-15:
Thanks.
I know change supported pins.

Why does not the sample code below work?

```

#include <Encoder.h>

Encoder myEnc(7, 8);

void setup() {
  Serial.begin(9600);
  Serial.println("Basic Encoder Test:");
}

long oldPosition  = -999;

void loop() {
  long newPosition = myEnc.read();
  if (newPosition != oldPosition) {
    oldPosition = newPosition;
    Serial.println(newPosition);
  }
}
```

DaisukeUra, 2017-06-15:
This code is Encoder/examples/Basic/Basic.pde.