PaulStoffregen/Encoder #21: #define INPUT_PULLUP /Arduino Due

Der-Modellmann, 2017-09-25:
If I #define INPUT_PULLUP an error occure.

Arduino: 1.8.2 (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from C:\Users\gabri\Google Drive\Arduino\Programme\________\achse5test.ino:12:0:

C:\Users\gabri\Google Drive\Arduino\libraries\Encoder-master/Encoder.h: In constructor 'Encoder::Encoder(uint8_t, uint8_t)':

C:\Users\gabri\Google Drive\Arduino\libraries\Encoder-master/Encoder.h:74:29: error: expected primary-expression before ')' token

   pinMode(pin1, INPUT_PULLUP);

                             ^

C:\Users\gabri\Google Drive\Arduino\libraries\Encoder-master/Encoder.h:75:29: error: expected primary-expression before ')' token

   pinMode(pin2, INPUT_PULLUP);

                             ^

exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

Is this my fault or an error in that library ?

Thanks
Gabriel

dagnall53, 2019-01-03:
INPUT_PULLUP is a #define normally "invisibly" included in arduino to allow you to code the pinMode with something recognizable like :     pinMode(EncoderPinA, INPUT_PULLUP);
Re defining it causes issues with the compiler.

PaulStoffregen, 2019-01-03:
Yes, you're correct.  This is an old, invalid issue.  Closing it now.

closed 2019-01-03 by PaulStoffregen