PaulStoffregen/Encoder #26: 4 steps in one

aster94, 2017-12-12:
hello Paul,

i m trying your library both on arduino uno and attiny167
for some reason i get 4 steps for every movements of the rotary encoder
the encoder is one of the cheap everyone use from aliexpress and i am not using any pull up/down/caps

to have your lib working i am using `myEnc.read()/4` do you know why is this happening?

cheers

PaulStoffregen, 2017-12-13:
This is normal.

closed 2017-12-13 by PaulStoffregen

bradanlane, 2018-12-12:
Given this is normal, and multiple users have encountered this issue, it may help to make this part of the reader, examples, Etc.

PaulStoffregen, 2018-12-12:
It's only normal for certain types of encoders, with detents or "clicky" feel.  Definitely not a generalization to all encoders, especially not ones used for sensing motor shaft or linear movement of machines.

bradanlane, 2018-12-12:
Gotcha. My installation is using _admittedly_ budget rotary encoders for a user interface.

Mesihas, 2020-12-02:
I am having the same issue but the encoder works fine when using the regular Encoder library with 32u4. I am wondering which kind of encoders can work for ESP32Encoder lib. regarding that 98% are from china.

1HighDuke, 2021-02-09:
> I am having the same issue but the encoder works fine when using the regular Encoder library with 32u4. I am wondering which kind of encoders can work for ESP32Encoder lib. regarding that 98% are from china.

Can confirm that on an Arduino Pro Micro (32u4) it counts multiples of 1 but on a Arduino Nano 168P it counts multiples of 4.

drf5n, 2021-04-19:
> Can confirm that on an Arduino Pro Micro (32u4) it counts multiples of 1 but on a Arduino Nano 168P it counts multiples of 4.

It is likely the detents interacting with the interrupts -- If the encoder library can use interrupts (32u4), it will, and will trigger on the rising and falling edges of the A and B signals.  If it cannot interrupt on the changes in the signal (168P), it resorts to polling (probably at the detents) and treats the A and B signals as "Clock" and "Data" and doesn't trigger at the pulse edges.  

There's 1 detent per cycle, but 4 pulse-edges per cycle.