PaulStoffregen/Encoder #27: ESP32 Not working.

ageir, 2017-12-14:
I can't get the library working on an ESP32 lolin. I see that ESP32 is in the .h files. But for some reason it's not working. Here's an excerpt of the errors.

ESP32 is defined, because if i try to define it it says I'm redefining it. No idea how to solve it. 

> 
> Compiling .pioenvs/lolin32/FrameworkArduino/HardwareSerial.o
> In file included from /home/myuser/.platformio/lib/Encoder_ID129/Encoder.h:46:0,
> from src/main.cpp:7:
> /home/myuser/.platformio/lib/Encoder_ID129/utility/interrupt_pins.h:202:2: error: #error "Interrupts are unknown for th
> is board, please add to this code"
> #error "Interrupts are unknown for this board, please add to this code"
> ^
> /home/myuser/.platformio/lib/Encoder_ID129/utility/interrupt_pins.h:205:2: error: #error "Encoder requires interrupt pi
> ns, but this board does not have any :("
> #error "Encoder requires interrupt pins, but this board does not have any :("
> ^
> ^
> /home/myuser/.platformio/lib/Encoder_ID129/utility/interrupt_pins.h:206:2: error: #error "You could try defining ENCODE
> R_DO_NOT_USE_INTERRUPTS as a kludge."
> #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."
> ^
> In file included from src/main.cpp:7:0:
> /home/myuser/.platformio/lib/Encoder_ID129/Encoder.h:61:11: error: 'IO_REG_TYPE' does not name a type
> volatile IO_REG_TYPE * pin1_register;
> ^
> /home/myuser/.platformio/lib/Encoder_ID129/Encoder.h:62:11: error: 'IO_REG_TYPE' does not name a type
> volatile IO_REG_TYPE * pin2_register;
> ^

PaulStoffregen, 2017-12-14:
I do not use ESP.  I'm depending on the ESP community to resolve problems and submit pull requests.

ageir, 2017-12-15:
Ah. Okej. Hopefully someone who knows the esp32 code sees this and can give me a hint on how to solve it. :)

Flos, 2018-01-13:
I had also trouble to compile the lib with my "nodemcu-32s" board. After looking at the source code, the ESP32 seems to be supported already. 
The switch for the interrupts is done with by checking for a define of "ESP32".  This define seems to be missing. After adding "#define ESP32" to my main.cpp the project compiles fine.

PaulStoffregen, 2018-01-14:
I'm depending on the ESP community to submit pull requests.  If someone knows the proper way to conditionally compile for ESP32, please submit a pull request.

Again, I do not use or test with ESP32.  This is free open source software.  I'm depending on you, the ESP community, to send good quality pull requests.

fredericplante, 2018-01-14:
I don't want to be a pain, but the esp32 is far from properly working. This will take time, as it took time for the esp8266, to become reliable.

Jim-McDonald, 2018-02-03:
I can confirm that this library works on an ESP32 (Lolin32 Lite) under Arduino IDE (with the Espressif core of course).

@ageir I see you are using PlatformIO. I've never used that IDE but suspect Espressif do *not* support it. I believe that is your trouble.

@fredericplante that has been true, especially six months ago but I'm finding that isn't really the case now (some edge cases of course), it's mostly generic libraries (or specific to AVR) and/or having not updated the Espressif core. Still, that isn't really on topic but thought I'd mention it in case other ESP32 users assume it's Paul's library or something.

Luke1962, 2018-05-24:
Hi, also on my side I can confirm that this libray is working properly on WEMOS LOLIN32 ( based on ESP32) , with 2 encoders.
My dev. environment is VS2015 + VisualMicro, based on Arduino IDE toolchain.

The only issue I've found , but not caused by this great library , is that if I choose some pins (e.g. 0 and 2 for EncA and 7,6 for EncB, the program becomes instable (infinite reset with message "Guru Meditation Error: Core  0 panic'ed (IllegalInstruction))". This does not happen if i choose pin 36,25 and 26,11.

euphy, 2018-06-08:
Going to add a little follow-up here because I couldn't understand how it worked for some ESP32 users and not me. Well, the ESP32 updates are _not_ in v1.4.1, and that is the most recent version available through library managers in Arduino IDE and PlatformIO.

I manually updated direct_pin_read.h and interrupt_pins.h from the head of this repo (utility folder) and now it compiles happily.

Brilliant work @PaulStoffregen, thank you to @lumostor and @aster94 for the ESP32 fixes!

Zefram88, 2019-07-27:
Tried and works on ESP32, but just on "No_interrupt" mode

majd47, 2019-10-18:
does esp32 works with pi 3 as a sender?

willemx, 2019-12-13:
I am also running into this problem. I can only get it to work by entering this dependancy in my `platformio.ini: lib_deps = https://github.com/PaulStoffregen/Encoder.git`
This will install version `#34ea332` of the library and this works without any problems.
If I enter `lib_deps = Encoder`, the latest version `#4c4ec3a` of the library gets installed and this will produce the errors again. Can someone explain this?

fercapa, 2020-02-15:
I was having the same problem in a laptop but not in a desktop computer. Both with same Arduino IDE and Encoder library version. In laptop I installed the library with Arduino tool and in desktop I did it manually from this github repository, and that was the trick.  After installed manually in laptop it compiled ok.

lumostor, 2020-02-16:
The version of encoder library from the arduino IDE and platformio is 1.4.1 (commit 4c4ec3a) which is prior ESP32 addition. You should install from github.

aidarishkulov, 2020-08-26:
The encoder does not work with ESP32!
Here is the correct library.
https://forum.flprog.ru/viewtopic.php?f=74&t=5073&p=92942#p92942

leckietech, 2021-09-13:
I am also using platformIO with no luck compiling for esp32dev. 

In file included from .pio/libdeps/esp32dev/Encoder/Encoder.h:46:0,
                 from src/main.cpp:14:
.pio/libdeps/esp32dev/Encoder/utility/interrupt_pins.h:202:2: error: #error "Interrupts are unknown for this board, please add to this code"
 #error "Interrupts are unknown for this board, please add to this code"
  ^
.pio/libdeps/esp32dev/Encoder/utility/interrupt_pins.h:205:2: error: #error "Encoder requires interrupt pins, but this board does not have any :("
 #error "Encoder requires interrupt pins, but this board does not have any :("
  ^
.pio/libdeps/esp32dev/Encoder/utility/interrupt_pins.h:206:2: error: #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."
 #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge." 

works flawlessly if I compile for esp32 using arduino IDE. I dont know how to define the board in pins.h, im a newbie.

RGeorge10, 2022-04-27:
There are solution for "dangerous relocation: l32r: literal placed after use:" error which help me with Arduino IDE.  I changed -ffunction-sections to -mtext-section-literals in C:\Users\*****\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\platform.txt file, in compiler.c.cmd=xtensa-esp32-elf-gcc and  compiler.cpp.cmd=xtensa-esp32-elf-g++ paragraphs. After changes two knobs encoder example compiles normal.