PaulStoffregen/Encoder #31: Compilation warning on no interrupt example

highvoltspam, 2018-11-19:
### Description

The no interrupt example throws out a bunch of errors on a new arduino 1.87 installation:
Encoder.h:69:7: Warning: type 'struct encoder' voilates one definition rule [-Wodr]
Encoder.h:69:7 - a different type is defined in another translation unit.
Encoder.h:132:10: the first difference ofr corresponding definitions is field 'interrupts_in_use'
etc...

### Steps To Reproduce Problem

Installed new Arduino 1.8.7 on win10 machine.
Installed Encoder library
compiled No interrupt example

PaulStoffregen, 2018-12-04:
Which board is selected in Tools > Boards?

Tolteque, 2020-11-15:
Hi,
  I have the same compilation issues (warnings) when compiling for "Arduino Micro"

  #define ENCODER_DO_NOT_USE_INTERRUPTS
  #include <Encoder.h>

  Have you find a solution ? Thx