PaulStoffregen/Encoder #40: Crashing on ESP8266 with core 2.5.2

ramdor, 2019-05-25:
### Description

Using the Basic Example from the Encoder Library, and ESP8266 core 2.5.2, the esp8266 crashes/resets.

### Steps To Reproduce Problem

Just use the Basic example included with the Encoder lib, and an ESP8266 with core version 2.5.2. I think same issue with 2.5.1, but ok with 2.5.0.

### Hardware & Software

Board : NodeMCU 1.0 (ESP-12E Module)
Board package : , version 2.5.2
Shields / modules used : none
Arduino IDE version : 1.8.9
Version info & package name (from Tools > Boards > Board Manager) : 1.4.1
Operating system & version : Win10

### Arduino Sketch

```cpp
/* Encoder Library - Basic Example
 * http://www.pjrc.com/teensy/td_libs_Encoder.html
 *
 * This example code is in the public domain.
 */

#include <Encoder.h>

// Change these two numbers to the pins connected to your encoder.
//   Best Performance: both pins have interrupt capability
//   Good Performance: only the first pin has interrupt capability
//   Low Performance:  neither pin has interrupt capability
Encoder myEnc(5, 4);
//   avoid using pins with LEDs attached

void setup() {
  Serial.begin(115200);
  Serial.println("Basic Encoder Test:");
}

long oldPosition  = -999;

void loop() {
  long newPosition = myEnc.read();
  if (newPosition != oldPosition) {
    oldPosition = newPosition;
    Serial.println(newPosition);
  }
}
```

### Errors or Incorrect Output
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

odudex, 2019-05-28:
Yes, happened to me to. Only works in my Wemos D1 mini with:
#define ENCODER_DO_NOT_USE_INTERRUPTS

odudex, 2019-05-28:
I verified that using external interrupts(all times) is crashing ESP8266's, but using "ICACHE_RAM_ATTR" attribute before interrupt handlers fix the problem. ex:
void ICACHE_RAM_ATTR buttonAction(){
 ...
}

odudex, 2019-06-01:
I managed to make it work with interrupts in ESP8266.
I'm using PlatformIO, in Atom, but should work in Arduino IDE too.
I uninstalled the library, and cloned it manually in my project's "lib" folder.
Than modified the "Encoder.h" file:
I just added ICACHE_RAM_ATTR in front of all isr_x_ functions, like:
static void isr0(void) { update(interruptArgs[0]); }
to
static void ICACHE_RAM_ATTR isr0(void) { update(interruptArgs[0]); }

orlo11, 2019-10-05:
Hello,
I added ICACHE_RAM_ATTR to every isrX, but after upload my WEMOS D1 mini still crashes with
"ISR not in IRAM!" massage. Did I forget something? I use 2M SPIFFS btw.

odudex, 2019-10-05:
> Hello,
> I added ICACHE_RAM_ATTR to every isrX, but after upload my WEMOS D1 mini still crashes with
> "ISR not in IRAM!" massage. Did I forget something? I use 2M SPIFFS btw.
To every isrX including your code and other libs?

orlo11, 2019-10-05:
to the isrX (X in 0..59) in Encoder.h
its a sketch to test so no other libs are included and no other isr is in the sketch itself.

orlo11, 2019-10-05:
Sorry I was not right there was another interrupt to handle the button.
But this fails to compile when I put ICACHE_RAM_ATTR in front of the handler ("not delared in this scope").

RLars, 2019-11-24:
I stumbled over the same problem. 
Created fix and pull request.

ALD136mtz, 2020-06-19:
Hi! I have the same issue with ESP8266 D1 mini, I downloaded the Encoder version 1.4.1, even before discovering this chat. I assumed that the library is updated, but keep getting the same issue. What can I do?

oferreiro, 2020-10-07:
> Hi! I have the same issue with ESP8266 D1 mini, I downloaded the Encoder version 1.4.1, even before discovering this chat. I assumed that the library is updated, but keep getting the same issue. What can I do?

Today, I only have success with this @PaulStoffregen version if interrupts is disabled.
```
#define ENCODER_DO_NOT_USE_INTERRUPTS
#include <Encoder.h>
```

With esp8266 chips, the @RLars version works well to me. Just delete your Encoder library and clone https://github.com/RLars/Encoder into your library directory.

closed 2021-06-27 by PaulStoffregen