PaulStoffregen/Encoder #44: Compile errors when using Teensy4

michaelmargolis, 2019-09-21:
### Description

Encoder example won't compile for Teensy4

### Steps To Reproduce Problem

select Teensy 4 and compile Basic example

### Hardware & Software

Teensy 4
Arduino IDE version 1.8.9
Teensyduino version 1.47
Windows 10

### Error

```
C:\Users\Michael\Documents\Arduino\libraries\Encoder/Encoder.h: In constructor 'Encoder::Encoder(uint8_t, uint8_t)':

C:\Users\Michael\Documents\Arduino\libraries\Encoder/Encoder.h:82:25: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

   encoder.pin1_register = PIN_TO_BASEREG(pin1);
```

maxgerhardt, 2019-11-05:
This is a side-effect that PlatformIO gives you an old version of the library because the lib author hasn't issued a version increase despite the critical commit https://github.com/PaulStoffregen/Encoder/commit/cfe65410a9d6a79fcea63a69ded7abccf281e752 which fixes the issue. In the `platformio` you should add  `https://github.com/PaulStoffregen/Encoder.git` to the `lib_deps` directive to use the updated version.

michaelmargolis, 2019-11-05:
The problem is because the reg macros are not defined for the teensy4 (__IMXRT1062__) in direct_pin_red.h
Adding the pin defines fixes this issue.
```
#if defined(__IMXRT1062__)
#define IO_REG_TYPE			    uint32_t
#define PIN_TO_BASEREG(pin)             (portInputRegister(digitalPinToPort(pin)))
#define PIN_TO_BITMASK(pin)             (digitalPinToBitMask(pin))
#define DIRECT_PIN_READ(base, mask)     (((*(base)) & (mask)) ? 1 : 0)
```