PaulStoffregen/Encoder #45: Counts by multiples of 2

rafununu, 2019-09-28:
It works pretty fine for me except that it returns pulses by pairs, so I can't stop the encoder on an impair value. That is for sure due to the cheap chinese encoders I use, otherwise I think somebody else would have report about that much before.

Anyway, this isn't a real issue, I simply divide the return number by 2 !

The Arduino librairie is still 1.41.

Thanks all for your work.

Cheers.

magic-mouse, 2019-10-26:
I can confirm this, except mine is multiplied by 4.

ondras12345, 2019-11-17:
Hi,
see https://github.com/PaulStoffregen/Encoder/issues/26 - this behavior is normal with certain types of rotary encoders.

Ipk4Fun, 2021-06-26:
> It works pretty fine for me except that it returns pulses by pairs, so I can't stop the encoder on an impair value. That is for sure due to the cheap chinese encoders I use, otherwise I think [...]

I pretty new to coding and can't figure out how to divide the return value, could you help me out?
Thanks!

PaulStoffregen, 2021-06-27:
myEnc.read() / 2;