PaulStoffregen/Encoder #50: Not working with Arduino Nano BLE 33

flocked, 2019-11-26:
### Description
Im getting error messages trying to use this library with the new Arduino Nano BLE 33.

### Steps To Reproduce Problem
See error message bellow

### Hardware & Software
Board
Arduino Nano BLE 33

Arduino IDE version
1.8.10

Version info & package name (from Tools > Boards > Board Manager)
Arduino Nano BLE 33

Operating system & version
Mac OS 10.15

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#define ENCODER_DO_NOT_USE_INTERRUPTS
#include <Encoder.h>
Encoder myEnc(5, 6);

void setup() {
  Serial.begin(9600);
  Serial.println("Basic NoInterrupts Test:");
}

long position  = -999;

void loop() {
  long newPos = myEnc.read();
  if (newPos != position) {
    position = newPos;
    Serial.println(position);
  }
}
```

### Errors or Incorrect Output
```
In file included from /Users/USERNAME/Documents/Arduino/libraries/Encoder/examples/NoInterrupts/NoInterrupts.pde:14:0:
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:61:11: error: 'IO_REG_TYPE' does not name a type
  volatile IO_REG_TYPE * pin1_register;
           ^~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:62:11: error: 'IO_REG_TYPE' does not name a type
  volatile IO_REG_TYPE * pin2_register;
           ^~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:63:2: error: 'IO_REG_TYPE' does not name a type
  IO_REG_TYPE            pin1_bitmask;
  ^~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:64:2: error: 'IO_REG_TYPE' does not name a type
  IO_REG_TYPE            pin2_bitmask;
  ^~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h: In constructor 'Encoder::Encoder(uint8_t, uint8_t)':
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:82:11: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'
   encoder.pin1_register = PIN_TO_BASEREG(pin1);
           ^~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:82:27: error: 'PIN_TO_BASEREG' was not declared in this scope
   encoder.pin1_register = PIN_TO_BASEREG(pin1);
                           ^~~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:82:27: note: suggested alternative: 'INIT_OBSERVER'
   encoder.pin1_register = PIN_TO_BASEREG(pin1);
                           ^~~~~~~~~~~~~~
                           INIT_OBSERVER
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:83:11: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'
   encoder.pin1_bitmask = PIN_TO_BITMASK(pin1);
           ^~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:83:26: error: 'PIN_TO_BITMASK' was not declared in this scope
   encoder.pin1_bitmask = PIN_TO_BITMASK(pin1);
                          ^~~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:83:26: note: suggested alternative: 'PIN_SPI_MOSI'
   encoder.pin1_bitmask = PIN_TO_BITMASK(pin1);
                          ^~~~~~~~~~~~~~
                          PIN_SPI_MOSI
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:84:11: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'
   encoder.pin2_register = PIN_TO_BASEREG(pin2);
           ^~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:85:11: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'
   encoder.pin2_bitmask = PIN_TO_BITMASK(pin2);
           ^~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:92:31: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'
   if (DIRECT_PIN_READ(encoder.pin1_register, encoder.pin1_bitmask)) s |= 1;
                               ^~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:92:54: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'
   if (DIRECT_PIN_READ(encoder.pin1_register, encoder.pin1_bitmask)) s |= 1;
                                                      ^~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:92:7: error: 'DIRECT_PIN_READ' was not declared in this scope
   if (DIRECT_PIN_READ(encoder.pin1_register, encoder.pin1_bitmask)) s |= 1;
       ^~~~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:93:31: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'
   if (DIRECT_PIN_READ(encoder.pin2_register, encoder.pin2_bitmask)) s |= 2;
                               ^~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:93:54: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'
   if (DIRECT_PIN_READ(encoder.pin2_register, encoder.pin2_bitmask)) s |= 2;
                                                      ^~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:93:7: error: 'DIRECT_PIN_READ' was not declared in this scope
   if (DIRECT_PIN_READ(encoder.pin2_register, encoder.pin2_bitmask)) s |= 2;
       ^~~~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h: In static member function 'static void Encoder::update(Encoder_internal_state_t*)':
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:274:40: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'
   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);
                                        ^~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:274:60: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'
   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);
                                                            ^~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:274:19: error: 'DIRECT_PIN_READ' was not declared in this scope
   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);
                   ^~~~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:275:40: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'
   uint8_t p2val = DIRECT_PIN_READ(arg->pin2_register, arg->pin2_bitmask);
                                        ^~~~~~~~~~~~~
/Users/USERNAME/Documents/Arduino/libraries/Encoder/Encoder.h:275:60: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'
   uint8_t p2val = DIRECT_PIN_READ(arg->pin2_register, arg->pin2_bitmask);
                                                            ^~~~~~~~~~~~
Mehrere Bibliotheken wurden für "Encoder.h" gefunden
 Benutzt: /Users/USERNAME/Documents/Arduino/libraries/Encoder
exit status 1
Fehler beim Kompilieren für das Board Arduino Nano 33 BLE.

```

f-jiang, 2020-04-25:
I'm having this issue as well. Have you made any progress on getting encoders to work on your model Arduino, whether with this library or not?

PaulStoffregen, 2020-04-27:
Contributions wanted.  Depending on the open source community to supply the required direct GPIO defines...

rantali, 2020-11-11:
Unfortunately the Encoder.h library in the nano BLE 33 doesn’t work..what can I do to make it work? I get "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge" and  "Encoder requires interrupt pins, but this board does not have any :(" but I know that BLE 33 have interrupts pins. Thank you very much for your help!!

PaulStoffregen, 2020-11-11:
Sorry, I can't offer direct technical support for porting to this board.  This is free open source software.  I'm depending on the open source community to contribute pull requests for this sort of thing.

The best I can tell you is to read these 3 header files and try to understand how defining details for boards works.

https://github.com/PaulStoffregen/Encoder/tree/master/utility

If you can't understand how those 3 files work, maybe try asking on the Arduino forum.  Perhaps you can find someone with the requite programming experience to understand the header files and deep knowledge of this board's hardware who may be willing to help.  As with any request on any forum, you can maximize your chance of success if you focus on 3 things:  1 - make a good first impression, 2 - give context about the problem, 3 - share specific details.  Arduino's forum is very busy.  It's difficult to stand out and make a good impression.  Showing a photo of hardware you've built and a screenshot of the error can help your request to draw attention among so many others which are text only.

kimaernoudt, 2020-12-25:
Has anyone seen any update on this ?
Thanks!

rantali, 2020-12-25:
Not me..

PR #71 (2021-06-27): Add support for Arduino Nano 33 BLE (NRF52840)
Fixes #50 for the case of interrupts enabled. I did not test with interrupts disabled.

Interested in any comments, I am particularly unsure if I used the appropriate flag for detecting the NRF52840 platform.
closed 2021-06-27 by PaulStoffregen