PaulStoffregen/Encoder #54: Not an issue - but...

phaleas, 2020-02-14:
I'm using your excellent library with an ESP32 - after doing so with Arduinos before. It seems, that you changed some files (ESP pins/interrupts) within the current version 1.4.1. These changes effect the users of ESP platform heavily, and many of them would not expect changes within a version they've already have. Please build a 1.4.2, so that ESP users get the latest changes - which are important.

I can confirm that the encoder runs like charm on an ESP32 with the latest changes.

Patrick-Thomas, 2021-02-19:
Yes I second this, please release a 1.4.2 version with the latest changes so we can import a ESP32 compatible version easily into Arduino and platformio

PaulStoffregen, 2021-03-01:
Done
https://github.com/PaulStoffregen/Encoder/releases/tag/1.4.2

closed 2021-03-01 by PaulStoffregen