PaulStoffregen/Encoder #59: ENCODER_OPTIMIZE_INTERRUPTS not working for ATmega4809

raindropmedia, 2020-08-14:
### Description

ENCODER_OPTIMIZE_INTERRUPTS do not work for ATmega4809 (Nano Every)

When I use the optimized source code, I no longer get any values for .read ();

it works without "ENCODER_OPTIMIZE_INTERRUPTS".

### Steps to reproduce the problem

Two 2400 step encoders on an Arduino Nano Every.
Latest encoder library from Github from August 9th, 2020

board
Arduino Nano Every

Arduino IDE version
1.8.10

Operating system & version
Mac OS 10.14.6

### Arduino Sketch
`
//#define ENCODER_OPTIMIZE_INTERRUPTS
#include <Encoder.h>
#include <Wire.h>

#define enc1P1  2
#define enc1P2  4
#define enc2P1  3
#define enc2P2  5
#define bit4   A0
#define bit3   A1
#define bit2   A2
#define bit1   A3
#define led     6

Encoder knobRot(enc1P1, enc1P2);
Encoder knobX(enc2P1, enc2P2);

typedef struct {
  int16_t x;
  int16_t rot;
}
B_t;
B_t data;

byte datasend = 0;
byte i2cAddress = 1;
boolean ledState = false;
byte ledblink;

void setup() {
  pinMode(bit1, INPUT_PULLUP);
  pinMode(bit2, INPUT_PULLUP);
  pinMode(bit3, INPUT_PULLUP);
  pinMode(bit4, INPUT_PULLUP);
  pinMode(led, OUTPUT);
  digitalWrite(led, HIGH);
  bitWrite(i2cAddress, 0, !digitalRead(bit1));
  bitWrite(i2cAddress, 1, !digitalRead(bit2));
  bitWrite(i2cAddress, 2, !digitalRead(bit3));
  bitWrite(i2cAddress, 3, !digitalRead(bit4));

  Wire.begin(i2cAddress);                // join i2c bus with address #8
  Wire.setClock(400000);
  Wire.onRequest(requestEvent); // register event
  data.x = 0;  // assign a value to the long number
  data.rot = 0;  // assign a value to the long number
  delay(1000);
  digitalWrite(led, LOW);
}

void loop() {
  int32_t newX, newRot, tempRot;
  newRot = knobRot.read();
  newX = knobX.read();
  if(newRot<0){
    tempRot = newRot % 2400;
    data.rot = 2400+tempRot;
  }else{
    data.rot = newRot % 2400;
  }
  data.x  = newX;
  Serial.print("absX= ");
  Serial.print(data.x);
  Serial.print(", absRot= ");
  Serial.print(data.rot);
  Serial.print(", absRot= ");
  Serial.println(newRot);
  //delay(1);
}

void requestEvent() {
  datasend = Wire.write ((byte *) &data, sizeof data);  /* sends data ti Master */
  //data.rot = 0;
  //data.x = 0;
}`

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

MonsterEnergy39, 2020-12-28:
I have the same issue, using 2 linear encoders.

Using "_//#define ENCODER_OPTIMIZE_INTERRUPTS_" works, but can sometimes miss steps in the encoder, while
using "_#define ENCODER_OPTIMIZE_INTERRUPTS_" does not give me any step signal.

gagarcr, 2021-02-17:
ENCODER_OPTIMIZE_INTERRUPTS is not working yet, but have a look at #66 as not every pin was using interrupts

ByteVenom, 2021-07-05:
Still running into this issue with the Nano Every. Has anyone come up with a workaround?