PaulStoffregen/Encoder #63: Basic example does NOT work reliably with KY040 mechanical switch rotary-encoder

StefanL38, 2020-12-11:
I tested this library with the example "basic" with a KY040 rotary-encoder. This type of encoder has mechanical switches.
This is the result that I get when I turn the encoder
17:39:09.750 -> Basic Encoder Test:
17:39:09.750 -> 0
17:39:09.750 -> 1
17:39:09.750 -> 0
17:39:09.750 -> 1
17:39:09.750 -> 0
17:39:09.750 -> 1
17:39:09.793 -> 2
17:39:09.793 -> 3
17:39:09.793 -> 2
17:39:09.793 -> 3
17:39:09.793 -> 2
17:39:09.793 -> 3

I guess this is due to not handling properly any inbetween-states and bouncing of a mechanical-rotary-switch can have
I tried NewConder from GFV  but get a compiler-error when compiling for Teensy 4.0
For other boards the NewEnocder from GVF works very good!
So Paul won't you improve your encoder-lib to the same level of high reliability?

drf5n, 2021-04-19:
Works fine for me.  Do you have reliable connections?

The README.md shows a test board with using 1K resistors and caps to electrically debounce the mechanical switches.   

I don't typically use the RC filtering, but haven't noticed problems.

StefanL38, 2021-04-19:
I'm just using the library NewEncoder
in NewEncoder debouncing is done by software and it even handles all situations of switch A/B open / closed at any position if you change rotationg-direction. This was the big problem. encoder jumps values depending on switch A/B opened or closed at the point were the rotation-direction changes. 
NewEncoder has userdefinable limits and additional functions

best regards Stefan