PaulStoffregen/Encoder #64: ESP32 compiles but it is not accurate

Mesihas, 2020-12-13:
There is no point on posting this on EXPRESSIF.COM as they don't offer support for this and they say to use pulses or whatever thing that there is nothing to do with this library and Arduino IDE.

For me compiles perfect. The problem is that for each encoder rotation (1 point) I get multiples values and most of the times is not clean in terms of sequential move like:

33
34
33
34
33
34
35
36

represents 1 move... How can accurately get a proper and unified value for every rotation point ?

danthegoodman1, 2021-03-29:
I have a similar issue, except no matter which direction I go I only get either 0 or 1:
```
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
1
0
```

brunnwart, 2021-05-02:
My experience is that this lib cannot be used für esp32. It just "looses" many counts (using interrupts by taking into account the ISR_ATTR topic) and thus it is absolutely far from precise and therefore useless.
Unfortunately I haven't found a working alternative so far.

jpswensen, 2021-06-10:
I was just looking at this today and I think I have a potential solution on the ESP32, but don't know how it would integrate well with this library. Using FreeRTOS (which is available in ArduinoIDE with the ESP32 core), you can designate a thread to run on a particular core. 

As a quick first test, I plan on kicking off a single thread on the second core whos only responsibility is to poll the various GPIO set up for encoders. It seems like a waste of a whole core, but should be faster than interrupts.

StefanL38, 2022-02-19:
With ESP32 or ESP8266 use NewEncoder from gfvalvo instead
https://github.com/gfvalvo/NewEncoder

best regards Stefan