PaulStoffregen/Encoder #69: Where can i find the processor defines, to add support for new boards?

Chick92, 2021-06-03:
This library is hands down the best encoder library.

I'm trying to add support for the raspberry pi pico and for the robotis OpenCR, both of which i seem to use every day of late. Try as i might though, i can't find where the processor is defined, to add it in to your library. E.g for an arduino Mega it woud be -

#elif defined(AVR_ATmega1280) || defined(AVR_ATmega2560)

Cheers.

Ben

PaulStoffregen, 2021-06-03:
here:
https://github.com/PaulStoffregen/Encoder/tree/master/utility

Chick92, 2021-06-03:
That's where they're programmatically identified in your library, but where is this definition actually made? E.g if i want to add support for another board, is it in boards.txt or something similar?

PaulStoffregen, 2021-06-03:
I don't know exactly where names like AVR_ATmega2560 are defined.  I've always imagined it's built into the compiler.  It might be in avr-libc header files, or somewhere else in the toolchain.  For all practical purposes, those are from the compiler.

oerkel47, 2021-08-02:
@Chick92 Have you made progress with support for the Pi Pico? I think I got the encoder library adapted properly but the readings are a bit off with arduinoCore-mbed. I think there must be some inaccuracies with the interrupts.. But doesn't work with https://github.com/earlephilhower/arduino-pico at all.

Check it out if you like: https://github.com/oerkel47/Encoder/tree/oerkel47-pico-support

Chick92, 2021-08-02:
@oerkel47 no I gave up with it. But, as for your problem, try declaring pinMode again after you attach the interrupt. There’s a bug in the mbed core, I’ve flagged it with them (on my phone so no link sadly) but that should solve your issue.

oerkel47, 2021-08-05:
@Chick92 thanks for your reply. I already read through the discussion you are refering to. Unfortunately that's not the problem. It works in principle, but it skips steps or counts wrong direction etc. So something is inaccurate with the registering of the interrupts is my guess.

ale-novo, 2022-11-07:
Hi, any luck on this? i have this exact same problem with skipping steps or count in the incorrect direction.

ale-novo, 2022-11-25:
ive tested  earlephilhower using the BASIC ino example but serial wont report back not sure what is happening.

oerkel47, 2022-11-26:
I have tried it too, still doesn't work. Polling works fine with both cores. At least for casual applications.