PaulStoffregen/Encoder #72: Support for STM32

mfouadthabet, 2021-09-08:
### Description

Tried to use the lib with STM32 bluepill , it seems that it is not yet supported
here is the build error: #error "Interrupts are unknown for this board, please add to this code"

For some reason this library is tagged as supporting STM32 in platformio documentation. 

Do you have plans to support it soon?
Is it a matter of only adding a new define segment in the interrupt_pins.h header file?

PaulStoffregen, 2021-09-08:
> Is it a matter of only adding a new define segment in the interrupt_pins.h header file?

Yes (or probably yes), just a matter of missing defines.  Help wanted from any STM32 experts to send a pull request.

Just to be clear, I do not have any STM32 boards.  I'm depending on the open source community to send pull requests to add the appropriate defines.

lucky62, 2024-05-11:
As I know in STM32F103C8 any GPIO pin can be used for interrupts.
Just it is not clear for me how to define pins.

For example this is defined for ESP8266:
[image]

The questions:
- what is the meaning of EXTERNAL_NUM_INTERRUPTS? (in which cases is used?)
- should the number xx in CORE_INTxx_PIN be from 0 to CORE_NUM_INTERRUPT?
- is the order of defined pins important?
- what is the meaning of pin number at the end? (is it number printed at board? or port-pin e.g. PB8 or what?)

lucky62, 2024-05-11:
ok, I created the definitions for BluePill according [this]( https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103C8T_F103CB(T-U)/variant_PILL_F103Cx.cpp) and seems encoder is working.

I am using PlatformIO (arduino framework). So I created the extra file with definitions and added it to build flags:
`build_flags =
    -include src/stm32f103C8_interrupt_pins.h`

I am not sure if it is 100% correct, but simply it is working. File is attached.
Note that I must use the pin names without underscore (like PB12, not PB_12 as are in the mentioned board variant file).
[stm32f103C8_interrupt_pins.zip](https://github.com/PaulStoffregen/Encoder/files/15282958/stm32f103C8_interrupt_pins.zip)