PaulStoffregen/Encoder #76: Demo Code count 1,2,3 if your encoder reads four steps per tick

StefanL38, 2022-02-19:
Please use this form only to report code defects or bugs.

**## NO!**
As long as GitHub is unable to provide

- an easy to find and
- step by step with **a lot of screenshots** tutorial how to use these complexe thing with forking and pull-requesting

So if you really need a bug-report: Here it is 
"GitHubs explanations how GitHub works as a whole thing is spreaded all over a lot of links always just mentioning a part of the things assuming the reader has already lot's of knowledge about Github 

I simply use the issues for this:

if your encoder reads four times for every mechanical tick of your encoder you can use this logic
[code]
#include <Encoder.h>  // von Paul Stoffregen

// If your encoder counts up by for for each step
// you can use this logic

const int CLK = 6;
const int DT = 5;

long altePosition = 0;
long neuePosition = 0;

long lastPos = 0;
long Zaehler = 0;
long alterZaehler = 0;

Encoder meinEncoder(DT, CLK);

void setup() {
  Serial.begin(115200);
} // SETUP

void loop() {
  neuePosition = meinEncoder.read();

  if (lastPos < neuePosition) {
    if (neuePosition % 4 == 0) {
      Zaehler++;
      lastPos = neuePosition;
    }
  }

  if (lastPos > neuePosition) {
    if (neuePosition % 4 == 0) {
      Zaehler--;
      lastPos = neuePosition;
    }
  }

  if (Zaehler != alterZaehler) {
    alterZaehler = Zaehler;
    Serial.println(Zaehler);
  } // IF

} // LOOP
[/code]

best regards Stefan

StefanL38, 2022-02-22:
Here is a demo-code that uses two encoders and functions that are handy to use.
As I'm a native german and this demo-code was for a german user it is guess what ? written in german

```#include <Encoder.h>  // von Paul Stoffregen

// selbsterklärende Namen vergleichsweise lang
// aber dafür eben SELBST-erklärend  Endung "_pin" 
// zeigt an es ist ein IO_pin
const int Step_CLK_pin = 3; 
const int Step_DT_pin  = 4;

const int Speed_CLK_pin = 6;
const int Speed_DT_pin  = 7;

long eingestellte_Steps = 0;
long eingestellte_Speed = 0;

Encoder Step_Eingabe (Step_DT_pin ,Step_CLK_pin);
Encoder Speed_Eingabe(Speed_DT_pin,Speed_CLK_pin);

void setup() {
  Serial.begin(115200);
} // SETUP

// ACHTUNG ! BESONDERHEIT!! das "&"-Zeichen  
// MUSS da stehen, weil der geänderte Variablenwert 
// nach "oben" zurückgegeben werden soll !
boolean Step_Geandert(long &p_StepZahl ) {

  boolean RueckgabeWert = false; // wenn NICHT gedreht wurde false zurückgebeb
  static long neuStep;
  static long lastStep;
 
  neuStep = Step_Eingabe.read();

  if (lastStep < neuStep) {
    if (neuStep % 4 == 0) {
      p_StepZahl++;
      lastStep = neuStep;
      RueckgabeWert = true; // wenn WIRKLICH gedreht wurde true zurückgeben
    }
  }

  if (lastStep > neuStep) {
    if (neuStep % 4 == 0) {
      p_StepZahl--;
      lastStep = neuStep;
      RueckgabeWert = true;
    }
  }
  return RueckgabeWert;
}

boolean Speed_Geandert(long &p_Speed ) {

  boolean RueckgabeWert = false; // wenn NICHT gedreht wurde false zurückgebeb
  static long neuSpeed;
  static long lastSpeed;
 
  neuSpeed = Speed_Eingabe.read();

  if (lastSpeed < neuSpeed) {
    if (neuSpeed % 4 == 0) {
      p_Speed++;
      lastSpeed = neuSpeed;
      RueckgabeWert = true; // wenn WIRKLICH gedreht wurde true zurückgeben
    }
  }

  if (lastSpeed > neuSpeed) {
    if (neuSpeed % 4 == 0) {
      p_Speed--;
      lastSpeed = neuSpeed;
      RueckgabeWert = true;
    }
  }
  return RueckgabeWert;
}

void loop() {

  // das hier muss sehr oftaufgerufen werden damit das PRG
  // mitbekommt ob an einem Encoder gedreht wurde
  if ( Step_Geandert (eingestellte_Steps)   ) {
    Serial.print("steps:");
    Serial.println(eingestellte_Steps);
  } 

  if ( Speed_Geandert (eingestellte_Speed)   ) {
    Serial.print("speed:");
    Serial.println(eingestellte_Speed);
  } 

}

/* Erklärung des Codes
Das "Umrechnen" von Flanken in Encoder-Ticks wurde in 
zwei functions "Step_Geandert"  "Speed_Geandert
gepackt. 

Diese functions geben true oder false zurück
Je nachdem ob wirklich am Encoder gedreht wurde oder nicht
Das kann man dann in If-Abfragen benutzen um nur dann eine Aktion
auszulösen wenn WIRKLICH am Encoder gedreht wurde

Es wurden zwei globale Variablen 
"eingestellte_Steps" und "eingestellte_Speed"  
definiert.

Diese Variablen werden vom Aufruf der functions
"Step_Geandert"  "Speed_Geandert"  aktualisiert
Diese Aktualisierung wird durch das "&"-Zeichen in der function 
Defintion erreicht
 */```
best regards Stefan

kng, 2022-04-21:
I also found out I have one of the 4-steps per mechanical click encoders.
This was my simple solution, add some check for value change etc if needed.
`enc_pos = myEnc.read() >> 2;  // divide down to mechanical steps if needed`
It would be nice if the lib supported some sort of robust divide down for these.