PaulStoffregen/Encoder #79: No support for Arduino nano rp2040 connect

balta1980, 2022-03-30:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.
No support for  Arduino nano rp2040 connect

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

try to compile the basic example will fail, it will says the board does not have interrupt pins try to use no interrupt example also fails.

### Hardware & Software

Board  Arduino nano rp2040 connect
Shields / modules used no
Arduino IDE version 1.8.19
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager) 3.0.0 for Mbed os nano
Operating system & version windows 11 22581.200
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

roman-rs, 2022-04-22:
same issue here.
at first it complained that the board doesn't have any interrupt pins defined, so I added some to the 
utility/interrupt_pins.h file, like:
// Arduino RP2040 connect                                                                                                                                                                                                                                                                
#elif defined(ARDUINO_NANO_RP2040_CONNECT)                                                                                                                                                                                                                                               
  #define CORE_NUM_INTERRUPT    8                                                                                                                                                                                                                                                        
  #define CORE_INT0_PIN 10                                                                                                                                                                                                                                                               
  #define CORE_INT1_PIN 11                                                                                                                                                                                                                                                               
  #define CORE_INT2_PIN 12                                                                                                                                                                                                                                                               
  #define CORE_INT3_PIN 13                                                                                                                                                                                                                                                               
  #define CORE_INT4_PIN 14                                                                                                                                                                                                                                                               
  #define CORE_INT5_PIN 15                                                                                                                                                                                                                                                               
  #define CORE_INT6_PIN 16                                                                                                                                                                                                                                                               
  #define CORE_INT7_PIN 17 

and now I've got another errors I have no idea how to deal with:
Arduino/libraries/Encoder/Encoder.h:61:11: error: 'IO_REG_TYPE' does not name a type
  volatile IO_REG_TYPE * pin1_register;
...
Arduino/libraries/Encoder/Encoder.h:64:2: error: 'IO_REG_TYPE' does not name a type
  IO_REG_TYPE            pin2_bitmask;
...
Arduino/libraries/Encoder/Encoder.h:82:27: error: 'PIN_TO_BASEREG' was not declared in this scope
   encoder.pin1_register = PIN_TO_BASEREG(pin1);
...

roman-rs, 2022-04-22:
It works fine to me.
please double check that you're using the latest library version.
check the header files, e.g. https://github.com/PaulStoffregen/Encoder/blob/master/utility/interrupt_pins.h
in the latest version in contains the block for rp2040 connect like:
// Arduino Nano RP2040 Connect
#elif defined(ARDUINO_NANO_RP2040_CONNECT)
  #define CORE_NUM_INTERRUPT	20
  #define CORE_INT0_PIN		0
  #define CORE_INT1_PIN		1
  #define CORE_INT2_PIN		2
  #define CORE_INT3_PIN		3
....

balta1980, 2022-04-22:
Thank you, I am going to check lib version and try again.

PaulStoffregen, 2022-06-17:
interrupt_pins.h was updated some time ago with defines for ARDUINO_NANO_RP2040_CONNECT.

Is this issue resolved?

kusnezow, 2022-10-22:
For me it works fine if: