PaulStoffregen/Encoder #82: Raspberry Pico (RP2040) support?

Pomax, 2022-09-22:
### Description

I tried running this with a Raspberry Pico using the "Arduino Mbed OS RP2040 (pico)" board setting, and got this error:

```
Arduino: 1.8.19 (Windows 10), Board: "Raspberry Pi Pico"
In file included from C:\Users\Pomax\Documents\Arduino\libraries\Encoder/Encoder.h:46:0,
                 from C:\Users\Pomax\Downloads\pico arduino\test\test.ino:1:
C:\Users\Pomax\Documents\Arduino\libraries\Encoder/utility/interrupt_pins.h:334:2: error: #error "Interrupts are unknown for this board, please add to this code"
 #error "Interrupts are unknown for this board, please add to this code"
  ^~~~~
C:\Users\Pomax\Documents\Arduino\libraries\Encoder/utility/interrupt_pins.h:337:2: error: #error "Encoder requires interrupt pins, but this board does not have any :("
 #error "Encoder requires interrupt pins, but this board does not have any :("
  ^~~~~
C:\Users\Pomax\Documents\Arduino\libraries\Encoder/utility/interrupt_pins.h:338:2: error: #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."
 #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."
  ^~~~~
exit status 1
Error compiling for board Raspberry Pi Pico.
```

### Steps To Reproduce Problem

Hook up a Raspberry Pico, pick the above-mentioned board, have a sketch that purely consists of `#include <Encoder.h>` and hit compile.

### Hardware & Software

Board: Raspberry Pico
Shields / modules used: none
Arduino IDE version: 1.8.19
~Teensyduino version (if using Teensy)~
Version info & package name (from Tools > Boards > Board Manager): BN: Raspberry Pi Pico, VID: 2e8a, PID: 00C0, SN: <redacted>
Operating system & version: win10 pro x64
~Any other software or hardware?~

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

kusnezow, 2022-10-22:
For me it works fine if:

1. I download the latest master branch from github (it seems there is no release with the rpi 2040 changes yet, 1.4.2 does not have them)
2. I manually define ARDUINO_NANO_RP2040_CONNECT as extra compiler flags. I have a rpi pico and this define is not set automatically.

After that it works like a charm.

ricardocolonia, 2022-10-22:
> Hi @kusnezow i am very new in the arduino world, i hgave tried your solution but i get other compilation errors, could you explain me how to manually define the extra compiler flags?
Thank you very much

Pomax, 2022-10-24:
@kusnezow sounds like you may have a solution that can be turned into a PR so that other folks don't have to go through that and can just use the Arduino IDE's built in "install this thing for me" =)

kusnezow, 2022-10-25:
@Pomax @ricardocolonia I am sorry, I don't use the Arduino IDE. I am using the Arduino Framework through Platform.io and there you can add extra compiler Flags in the platformio.ini file (documented here: https://docs.platformio.org/en/stable/projectconf/section_env_build.html#scopes-scons-variables). Not sure how to achieve that in the Arduino IDE

PaulStoffregen, 2022-10-25:
I've published version 1.4.3.  Soon Arduino's Library Manager will notice the new version and this problem will be a thing of the past.

Pomax, 2022-10-25:
sweet! Many thanks

closed 2022-10-25 by PaulStoffregen