PaulStoffregen/Encoder #84: RP Pico RP2040 ConfigurableFirmata Encoder Interrupts not working

ale-novo, 2022-11-06:
Hello! been using ConfigurableFirmata V2.10.1 based from Firmatabuilder.com sketches successfully with Arduino.

Ive added support for RP2040 how is described here:
https://github.com/firmata/ConfigurableFirmata/blob/3a1747953150b6589b7d5bf1f3aa92c5aca3c365/BoardSupport.md

Everything works fine however ive notice that if i plug in an encoder it does not work accurately like it does in the Arduino nano port 2/3 that are interrupts pins.

In theory all RP2040 digital pins are interrupts but they are not behaving like interrupt pins but like normal pins therefore making the encoder not read accurately.

ive modified the encoder library here:
https://github.com/PaulStoffregen/Encoder/search?q=RP2040

to use 
#elif defined(TARGET_RP2040) || defined(TARGET_RASPBERRY_PI_PICO)
instead of
#elif defined(ARDUINO_NANO_RP2040_CONNECT)

Its compiling fine however not working accurately, please help!! thanks

RussCoty, 2023-03-08:
Did you get anywhere on this @ale-novo ? I am struggling to even get things compiling.

ale-novo, 2023-03-11:
@RussCoty i had other things so i did no further testing. The problem for me is happening with a Raw mechanical encoder directly connected, however i had some discussions with other people that successfully have a working encoder with RPI pico and they are using the rotary encoder ky-040 that is mechanical but has some discreet components. I have purchased one but still have to test it. Also for my application i actually use a optical encoder that has no 'bounce' problems so i will have to test with that as well.

ale-novo, 2023-03-13:
Hi, ive tested this with RP2040 and using the "examples/Basic/Basic.ino" sketch.
with a simple mechanical encoder connecting the common to ground and A, B to interrupt pins, works very unreliable.
with a KY-040 encoder connecting 5v, gnd and a,b to interrupt pins works better but still misses or counts incorrectly.
with a optical encoder did not test but i suspect that there is a problem with interrupts

all encoders work perfectly with arduino nano pins 2,3 (only interrupt pins)

RussCoty, 2023-03-22:
Thanks @ale-novo I have tried the modifications to encoder.h that you suggested but not even getting a successful compile 🤔.

I will post some actual error messages into this thread as soon as I can. 

I'm wondering if we could use the state machines to do this more robustly but I do think Paul's library is elegant and works like a dream on teensy and arduino.