PaulStoffregen/Encoder #89: library.properties Still Says Version 1.4.2

abrahamjsimpson, 2023-01-31:
library.properties has not been updated to reflect the release of version 1.4.3. Line 2 still says 
`version=1.4.2`
This should be updated to reflect the current version (1.4.3). As a result of this, the Arduino IDE says that the version right now is 1.4.2.

abrahamjsimpson, 2023-01-31:
Just noticed that ee9543ed91e294956031d3a86c4f93eef679d471 . Has this really been a problem for that long?

TadyTheFish, 2023-02-26:
I'm also trying to use the 1.4.3 version in Platformio and it does not exist because of this. I can only load 1.4.2 that does not have the Rap2040 support fixed

cattledogGH, 2023-03-29:
Getting an update to version 1.4.3 in the Arduino library manager is important because the previous versions have library examples with .pde files, and the latest 2.0.x ide will only open files with .ino extensions.

[https://github.com/arduino/arduino-ide/issues/518](url)

PaulStoffregen, 2023-03-29:
Hopefully this will solve the problem
https://github.com/PaulStoffregen/Encoder/commit/547fd926261dbdc87eee5b947ef30b4c63993cd3

cattledogGH, 2023-03-31:
The Arduino IDE library manager is still not showing V1.4.3.  

[https://github.com/arduino/library-registry/blob/main/FAQ.md#why-arent-releases-of-my-library-being-picked-up-by-library-manager](url)

bschippers, 2023-05-01:
The tag of v1.4.3 still contains library.properties saying it is v1.4.2. I think that's why it's not getting picked up. Can the tag be moved or make a new release? Here's the log from https://downloads.arduino.cc/libraries/logs/github.com/PaulStoffregen/Encoder/

2023/05/01 18:00:59 Scraping https://github.com/PaulStoffregen/Encoder.git
2023/05/01 18:00:59 Checking out tag: 1.3
2023/05/01 18:00:59 Release Encoder:1.3.0 already loaded, skipping
2023/05/01 18:00:59 Checking out tag: 1.4
2023/05/01 18:00:59 Release Encoder:1.4.0 already loaded, skipping
2023/05/01 18:00:59 Checking out tag: 1.4.1
2023/05/01 18:00:59 Release Encoder:1.4.1 already loaded, skipping
2023/05/01 18:00:59 Checking out tag: 1.4.2
2023/05/01 18:00:59 Release Encoder:1.4.2 already loaded, skipping
2023/05/01 18:00:59 Checking out tag: 1.4.3
2023/05/01 18:00:59 Release Encoder:1.4.2 already loaded, skipping

ivankravets, 2023-11-27:
The same issue with https://registry.platformio.org/libraries/paulstoffregen/Encoder

@PaulStoffregen, could you remove existing https://github.com/PaulStoffregen/Encoder/releases/tag/1.4.3 tag and create the same `1.4.3` from `547fd926261dbdc87eee5b947ef30b4c63993cd3` commit?

ivankravets, 2023-11-27:
The same issue with the `MCP_CAN_lib ` library by @coryjfowler and the `1.5.1` tag that points to the previous `1.5.0` version.

coryjfowler, 2023-11-28:
Why on earth am I being notified about this here????  A professional response would be linking me to a detailed process on how to do this correctly.  Instead 4 different people are just saying what the problem is and potentially one person that actually mentioned what to do to correct the problem for *this* repo.  I still don't know why I am being tagged here what so ever.

coryjfowler, 2023-11-28:
I do this in my absolute free time, so its cool I get to waste it commenting here instead of implementing a solution.  If anyone knows an employer who is closer to my house in Columbus Indiana than a 100 mile to work and a 100 mile drive back home, that would be great.

coryjfowler, 2023-11-28:
It looks like I released 1.5.1 before I updated library.properties.  The contents of the ZIP are still 1.5.1 regardless.  Is this non-issue an actual issue for someone or something?  I have other things I could be doing that trying to figure out why I am being tagged here.

coryjfowler, 2023-11-28:
Also this 1.5.1 tag points to https://github.com/coryjfowler/MCP_CAN_lib/releases/tag/1.5.1  1.5.1 ... I'm lost.  Other than that one digit in a text file in the zip.......

ivankravets, 2023-11-28:
@coryjfowler, sorry for disturbing you. You are in good company here, all of us love open source and invest our valuable time in it.

> A professional response would be linking me to a detailed process
It was provided in my original comment  https://github.com/PaulStoffregen/Encoder/issues/89#issuecomment-1827993230

> Also this 1.5.1 tag points

Could you open the [1.5.1 tag](https://github.com/coryjfowler/MCP_CAN_lib/tree/1.5.1) and click on `library.properties`. Which `version` value do you see?

---------

Solution. Please remove [1.5.1](https://github.com/coryjfowler/MCP_CAN_lib/releases) release and tag. Make a new one from the latest commit 6f1f30ce41e11bd75a8e367b7666399da8ddeea0 ( https://github.com/coryjfowler/MCP_CAN_lib/commits/master )

coryjfowler, 2023-11-29:
> It looks like I released 1.5.1 before I updated library.properties.
The contents of the source files should be 1.5.1 despite that.  I am absolutely not sure how to fix it, my git-fu is null and every time I try to learn it, it seems more convoluted and pointless than the last time I tried.  I edit via the web interface! LOL

coryjfowler, 2023-11-29:
@ivankravets  That horizontal bar made me want to interpret the contents below it as a forum signature at first read and I ignored it subconsciously.  Ha!

ivankravets, 2023-11-29:
Thanks, @coryjfowler! 🙏 

The library has been automatically updated in the PlatformIO Registry https://registry.platformio.org/libraries/coryjfowler/mcp_can

PaulStoffregen, 2023-12-11:
I edited library.properties to 1.4.4 and released a new 1.4.4 tag.

Hopefully that fully resolves this issue?

ivankravets, 2023-12-11:
Thanks, @PaulStoffregen. It works now! https://registry.platformio.org/libraries/paulstoffregen/Encoder

The issue can be closed.

closed 2023-12-11 by PaulStoffregen