PaulStoffregen/Encoder #90: Encoder causes esp32 to crash when using platformIO

BrentRoberson, 2023-02-22:
Used the example code and it works fine with the Arduino IDE but as soon as I take the exact code the PlatformIO my esp32 crashes 
and gives an error on repeat.

### Steps To Reproduce Problem
use the library in platformIO with the example given.

### Hardware & Software

Using Esp32 Development kit, latest version of encoder library 
Here is my Platformio.ini: 
[env]
build_src_filter = +<*> -<.git/> -<.svn/> -<example/> -<examples/> -<test/> -<tests/>

[env:dispenser]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
	miguelbalboa/MFRC522@^1.4.10
	marcoschwartz/LiquidCrystal_I2C@^1.1.4
	soligen2010/ClickEncoder@0.0.0-alpha+sha.9337a0c46c
	paulstoffregen/Encoder@^1.4.2
upload_port = COM8
monitor_port = COM8
monitor_speed = 115200
build_src_filter = ${env.src_filter} -<mainRegister.cpp> 

### Errors or Incorrect Output

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
E (166) gpio: esp_ipc_call_blocking failed (0x103)
[     6][E][esp32-hal-gpio.c:175] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start
E (169) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400f7467  PS      : 0x00060533  A0      : 0x800d6beb  A1      : 0x3ffe3b30  
A2      : 0x00000000  A3      : 0x3ffc2e60  A4      : 0x3ffc2e60  A5      : 0x00060523
A6      : 0x00060520  A7      : 0x00000001  A8      : 0x800d67a8  A9      : 0x3ffe3af0  
A10     : 0x3ffbdc14  A11     : 0x00000010  A12     : 0x00000004  A13     : 0x3ffe3b30
A14     : 0x007bdc00  A15     : 0x003fffff  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40086979  LEND    : 0x40086989  LCOUNT  : 0xfffffff7
Backtrace: 0x400f7464:0x3ffe3b30 0x400d6be8:0x3ffe3b50 0x400d4490:0x3ffe3b80 0x400d44dd:0x3ffe3bb0 0x400d1ed1:0x3ffe3bd0 0x400d25da:0x3ffe3bf0 0x400ddf1f:0x3ffe3c10 0x40082d71:0x3ffe3c40 0x400792ba:0x3ffe3c90 |<-CORRUPTED

tko, 2023-04-22:
[Comment on reddit](https://www.reddit.com/r/esp32/comments/yalvd6/attachinterrupt_has_stopped_working/itcq3re/) suggests this is due to attaching to the interrupts in the constructor and it might work if it were done in `setup()` instead. Unfortunately that doesn't seem to be possible with the current code.

ZephyrCloudNine, 2024-02-17:
Attaching the interrupt in void setup() is the way to go instead of at the constructor. Here's how i went about it given that the Encoder library requires you to pass the interrupt pins in the constructor itself. Heres an example code which worked for me (ESP32S dev module)

/* Encoder Library - Basic Example
 * http://www.pjrc.com/teensy/td_libs_Encoder.html
 *
 * This example code is in the public domain.
 */

#include <Encoder.h>

// Change these two numbers to the pins connected to your encoder.
//   Best Performance: both pins have interrupt capability
//   Good Performance: only the first pin has interrupt capability
//   Low Performance:  neither pin has interrupt capability
Encoder *myEnc;
//   avoid using pins with LEDs attached

void setup() {

  myEnc = new Encoder(18,19);
  Serial.begin(115200);
  Serial.println("Basic Encoder Test:");
}

long oldPosition  = -999;

void loop() {
  long newPosition = myEnc->read();
  if (newPosition != oldPosition) {
    oldPosition = newPosition;
    Serial.println(newPosition);
  }
}