PaulStoffregen/Encoder #93: Feature: getDirection()

dicagno, 2023-05-06:
### Description

A method to determine step direction is missing and would be a great nice-to-have.

### Steps To Reproduce Problem

No method to get step direction.

### Hardware & Software

Board: any
Shields / modules used: any

### Usage
```cpp
int direction = encoder.getDirection();
// 1 -> forward
// -1 -> backwards
// 0 -> no movement
```

### Errors or Incorrect Output

--

bwalc0, 2023-12-13:
If you would like to use it, I have edited **Examples** -> **Basic**

```
long oldPosition  = -999;

void loop() {
  long newPosition = myEnc.read();

  int direction = 0;
  if (newPosition != oldPosition) { 

    direction = (newPosition > oldPosition) ? 1 : -1;

    oldPosition = newPosition;
    Serial.println(newPosition);
  }
}
```

dicagno, 2023-12-17:
@bwalc0 thanks for the answer. Did you take a look at the associated PR?