PaulStoffregen/Encoder #95: Raspberry pi pico RP2040 Not supported

RichieUnpar, 2023-07-03:
### Description

i want to use this library using the rp2040 board but it seems the library does not directly support the raspberry pi version but it does support the Arduino Nano Rp2040 Connect. After adding the rp2040 board to the interrupt_pins.h with my text being:

//RP2040 Connect https://arduino-pico.readthedocs.io/en/latest/contrib.html#adding-a-new-board
#elif defined(ARDUINO_ARCH_RP2040) && !defined(__MBED__)
  #define CORE_NUM_INTERRUPT	26
  #define CORE_INT0_PIN		0
  #define CORE_INT1_PIN		1
  #define CORE_INT2_PIN		2
  #define CORE_INT3_PIN		3
  #define CORE_INT4_PIN		4
  #define CORE_INT5_PIN		5
  #define CORE_INT6_PIN		6
  #define CORE_INT7_PIN		7
  #define CORE_INT8_PIN		8
  #define CORE_INT9_PIN		9
  #define CORE_INT10_PIN	10
  #define CORE_INT11_PIN	11
  #define CORE_INT12_PIN	12
  #define CORE_INT13_PIN	13
  #define CORE_INT14_PIN	14
  #define CORE_INT15_PIN	15
  #define CORE_INT16_PIN	16
  #define CORE_INT17_PIN	17
  #define CORE_INT18_PIN	18
  #define CORE_INT19_PIN	19
  #define CORE_INT20_PIN	20
  #define CORE_INT21_PIN	21
  #define CORE_INT22_PIN	22
  #define CORE_INT23_PIN	23
  #define CORE_INT24_PIN	24
  #define CORE_INT25_PIN	25

### Steps To Reproduce Problem

upload any program that has the #include <Encoder.h> in the code
this problem persist whethever #define ENCODER_DO_NOT_USE_INTERRUPTS is used or not

### Hardware & Software

Board
Generic RP2040/ Raspberry pi pico rp2040 board

### Arduino Sketch

Any Example library will reproduce the problem, just include the  #include <Encoder.h>

### Errors or Incorrect Output

Arduino: 1.8.19 (Windows 10), Board: "Generic RP2040, 16MB (Sketch: 8MB, FS: 8MB), 240 MHz (Overclock), Small (-Os) (standard), Disabled, Disabled, Disabled, Disabled, None, Pico SDK, IPv4 Only, Generic SPI /2, Default (UF2)"

In file included from C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3\examples\Basic\Basic.ino:7:

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:68:11: error: 'IO_REG_TYPE' does not name a type; did you mean 'IP_GET_TYPE'?

   68 |  volatile IO_REG_TYPE * pin1_register;

      |           ^~~~~~~~~~~

      |           IP_GET_TYPE

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:69:11: error: 'IO_REG_TYPE' does not name a type; did you mean 'IP_GET_TYPE'?

   69 |  volatile IO_REG_TYPE * pin2_register;

      |           ^~~~~~~~~~~

      |           IP_GET_TYPE

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:70:2: error: 'IO_REG_TYPE' does not name a type; did you mean 'IP_GET_TYPE'?

   70 |  IO_REG_TYPE            pin1_bitmask;

      |  ^~~~~~~~~~~

      |  IP_GET_TYPE

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:71:2: error: 'IO_REG_TYPE' does not name a type; did you mean 'IP_GET_TYPE'?

   71 |  IO_REG_TYPE            pin2_bitmask;

      |  ^~~~~~~~~~~

      |  IP_GET_TYPE

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h: In constructor 'Encoder::Encoder(uint8_t, uint8_t)':

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:89:11: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'

   89 |   encoder.pin1_register = PIN_TO_BASEREG(pin1);

      |           ^~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:89:27: error: 'PIN_TO_BASEREG' was not declared in this scope

   89 |   encoder.pin1_register = PIN_TO_BASEREG(pin1);

      |                           ^~~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:90:11: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'

   90 |   encoder.pin1_bitmask = PIN_TO_BITMASK(pin1);

      |           ^~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:90:26: error: 'PIN_TO_BITMASK' was not declared in this scope

   90 |   encoder.pin1_bitmask = PIN_TO_BITMASK(pin1);

      |                          ^~~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:91:11: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'

   91 |   encoder.pin2_register = PIN_TO_BASEREG(pin2);

      |           ^~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:92:11: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'

   92 |   encoder.pin2_bitmask = PIN_TO_BITMASK(pin2);

      |           ^~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:99:31: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'

   99 |   if (DIRECT_PIN_READ(encoder.pin1_register, encoder.pin1_bitmask)) s |= 1;

      |                               ^~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:99:54: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'

   99 |   if (DIRECT_PIN_READ(encoder.pin1_register, encoder.pin1_bitmask)) s |= 1;

      |                                                      ^~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:99:7: error: 'DIRECT_PIN_READ' was not declared in this scope

   99 |   if (DIRECT_PIN_READ(encoder.pin1_register, encoder.pin1_bitmask)) s |= 1;

      |       ^~~~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:100:31: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'

  100 |   if (DIRECT_PIN_READ(encoder.pin2_register, encoder.pin2_bitmask)) s |= 2;

      |                               ^~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:100:54: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'

  100 |   if (DIRECT_PIN_READ(encoder.pin2_register, encoder.pin2_bitmask)) s |= 2;

      |                                                      ^~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:100:7: error: 'DIRECT_PIN_READ' was not declared in this scope

  100 |   if (DIRECT_PIN_READ(encoder.pin2_register, encoder.pin2_bitmask)) s |= 2;

      |       ^~~~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h: In static member function 'static void Encoder::update(Encoder_internal_state_t*)':

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:303:40: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'

  303 |   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);

      |                                        ^~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:303:60: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'

  303 |   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);

      |                                                            ^~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:303:19: error: 'DIRECT_PIN_READ' was not declared in this scope

  303 |   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);

      |                   ^~~~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:304:40: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'

  304 |   uint8_t p2val = DIRECT_PIN_READ(arg->pin2_register, arg->pin2_bitmask);

      |                                        ^~~~~~~~~~~~~

C:\Users\xxxxxxx\Documents\Arduino\libraries\Encoder-1.4.3/Encoder.h:304:60: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'

  304 |   uint8_t p2val = DIRECT_PIN_READ(arg->pin2_register, arg->pin2_bitmask);

      |                                                            ^~~~~~~~~~~~

exit status 1

Error compiling for board Generic RP2040.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mattordev, 2024-05-28:
Did you ever find a way to fix this?

MrPatben8, 2024-08-15:
There is still no support for the Pico despite the RP2040 being supported.

doctea, 2024-09-15:
I forget the details, so this shouldn't be considered a complete fix; but I use this fork successfully with my RP2040 project (along with initialising at runtime instead of statically, as discussed in #85): https://github.com/doctea/Encoder

neilmendoza, 2025-09-13:
@PaulStoffregen Most of the forks of this repo are adding support for the RP2040 and RP2350. Any chance of getting the changes in one of them, e.g. [this one](url), merged? Thanks!