PaulStoffregen/Encoder #98: Arduino UNO R4 MINIMA not supported

GralfR, 2023-09-17:
### Description

This library is not working on Arduino UNO R4 Minima. Lots of errors like "...has no member named..." concerning pin bitmask and registers.

### Steps To Reproduce Problem

Download library (seams to be version 1.4.3 on Sept 17 2023).
Add library to IDE (used 1.8.10).
Open Basic example.
Choose Arduino UNO R4 Minima as board.
Compile.

### Hardware & Software

Board: Arduino UNO R4 Minima
Shields / modules used: none
Arduino IDE version: 1.8.10
Operating system & version: Win10 pro

### Arduino Sketch

the included example "Basic".

### Errors or Incorrect Output

'IO_REG_TYPE' does not name a type

mich-j, 2024-02-06:
You can check out a [fork made by mjs513](https://github.com/mjs513/Encoder/tree/Arduino-R4), branch `Arduino-R4`. It works well with the R4.

pandel, 2024-03-11:
I tried the [for by mjs53](https://github.com/mjs513/Encoder/tree/Arduino-R4) today on a R4 WIFI, but sadly, nothing works. The encoder works absolutely flawless on a R3 with the original library, but not with the fork on the R4 and, for sure, not with the original library.

pandel, 2024-03-12:
Many thanks! It might be a solution for another project. I decided to drop the whole library and handle the encoder myself. Works beautiful so far so that I don't need any library atm.

tpwade, 2025-04-04:
Just an update with respect to the encoder library not working on the R4.  The library bundled with the Arduino IDE still seems to be outdated. The necessary ARDUINO_UNOR4_MINIMA and ARDUINO_UNOR4_WIFI defines are still missing in utility/direct_pin_read.h and utility/interrupt_pins.h in IDE supplied library. The reported "versions" do not necessarily seem correlate to which is "fixed".  These issues lead to a compiler error and can be fixed buy downloading and switching to the latest github repository which has the necessary defines.

Secondly there is still some bug with respect to the use of interrupt pins. According to the documentation optimum performance is achieved if both signals are attached to interrupt pins (eg Encoder myEnc(2, 3) in Basic.ino).  This works on an Arduino Nano where pins 2 and 3 are the physical interrupt pins, but fails on an Arduino Uno R4 MINIMA.  myEnc(2, 8) and myEnc(3, 8) also fail on the R4.  So long as the the second pin is not a physical interrupt things do seem to work on the R4 (eg myEnc(2,4) or myEnc(3,5)).